qba667/FlySkyI6 #99: WIiki - Missing ASL text

llcool, 2020-05-16:
Altitude calculation with TGY-CAT01.
System setup -> Extra -> ASL
Long press OK - set default value
Short press OK change cursor between temperature at see level and pressure at see level.
If desired value is altitude relative to start position change values to trigger:
-pressure equals actual pressure
-temperature equals actual pressure
Altitude should be re-calculated on every value change.

Yenya, 2020-05-28:
@llcool, thanks for your input. I tried to reword the text a bit (yet my English is not perfect, so corrections are welcome). Is it OK now?

https://github.com/qba667/FlySkyI6/wiki/ASL

llcool, 2020-05-28:
Hi Yenya,

Nice work. :)

I would word it like this:-

Above Sea Level (ASL) Sensor

The pressure sensor, such as TGY-CAT01, can be configured in the System ⭢
Extra ⭢ ASL menu:

The sensor cannot compute the altitude by itself, so the computation is
done inside the Tx firmware.
Setup

The *default values* can be set by long pressing the OK button.

A short press of the OK button is used to change the between the pressure at
the sea level and temperature. The a Altitude is calculated as altitude *above
the sea level*.

If the desired value is altitude *relative to start position* To set
the *relative
altitude, *set the pressure and temperature a current pressure and
temperature at the start position.

The altitude should be Altitude is re-calculated on every value change.

All the best,

Lloyd

Yenya, 2020-06-02:
@llcool: another try. Is it correct now? I had a bit of trouble decoding which parts of text do you want me to modify. I hope I understood you well.

closed 2020-06-03 by llcool