qba667/FlySkyI6 #101: No SWE found after update FLYPLUS 1.7.6

louisFpv, 2020-05-27:
Thx your contribution for the new firmware 1.7.6.

I have updated the new firmware 1.7.6 but no swe found in aux channels no matter which bin file fs-i6-swe_updater_05_07_12_30, or fs-i6_updater_05_07_12_30 installed. My i6 was already updated to 1.7.5 before and mod swb and swe, it works ok and good. 
Would you like to check it whether the SWE bin file is missed or right?

Thx & regards

Yenya, 2020-05-27:
@louisFpv, can you test the attached firmware file?
[swe-test.zip](https://github.com/qba667/FlySkyI6/files/4689465/swe-test.zip)
It might be that the firmware got built incorrectly.

louisFpv, 2020-05-27:
Thx, SWE presents again. Well Done

[image]
[image]

Yenya, 2020-05-28:
OK, thanks for the bug report and for testing this. I have replaced the SwE firmware in the 1.7.6 relase zip archive with the above one.

closed 2020-05-28 by Yenya