qba667/FlySkyI6 #102: Reverse function doesn't  work.

bladyle, 2020-06-01:
I have set Failsafe on channel 3 and SwA is assigned as Throttle hold on channel 5 so it's working OK but I want Throttle hold to be engaged on up position and I reverse channel 5 but it wont change how SwA is working.  It tried this with SwB which is set as channel 6 and then reversed but result was the same - function doesn't work.

Yenya, 2020-06-01:
This is a feature, not a bug. It behaves the same in the original FlySky firmware, and other special switches assignments (Teacher/Student, Throttle curve, and Dual rates) work the same way - they are not affected by reversing of the given channel. In fact, the switches do theirs special function even without being assigned to a channel at all.

bladyle, 2020-06-02:
I'm flying only quads so for me I can't see as feature but maybe for planes it is.  I'm pretty sure that is not like this in original firmware because I have some memory of using my arm switch reversed, but I can be wrong. If I have some free time I can flash back original and test it. It's not a big problem for me, so everything is OK if that is expected behavior .

Yenya, 2020-06-02:
I have one FS-i6 with stock firmware - I have verified in the Key->Display menu, that when I set up SwA as Throttle hold in the Switches assign menu, the throttle hold on Ch3 happens when I pull the SwA to the down position (towards myself), and in the default position, which is checked after power-on, it is not engaged, i.e the throttle stick works and moves the Ch3 value. This is regardless of assignment of SwA to any particular channel - it happens when I don't have SwA assigned at all, it happens when I assign SwA to Ch6, and it also happens when I reverse the Ch6 in the Mix->Reverse menu.

That said - for quads, it might be that the arm switch is not handled by the Tx itself, but by the flight controller inside the quad. Then the switch of course needs to be sent to the Rx as a channel (reversed or not), and probably the Throttle hold function inside the Tx should be disabled or at least assigned to another switch.

bladyle, 2020-06-02:
You can flash original firmware by downloading it from official site https://www.flysky-cn.com/i6-xiazai
Another thing I can't remember is how was it on original firmware -  in Switch assign menu there is no option mode to be empty or assigned to NONE. I saw that on FS-i6x original version this is posible.

Yenya, 2020-06-02:
Thanks. It is apparently a Windows EXE file - I'll need to figure out how to unpack it under Linux first. Also, it is labeled "2.0.7" - my FS-i6 with stock firmware has firmware labeled "1.1 27-Mar-2015".

Yes, in the stock firmware in the Swtiches assign menu, there is no option to set the function to None. Only SwA, SwB, or SwD. It is the same in the FlyPlus firmware.