qba667/FlySkyI6 #103: [NeedInfo] Is more channel could be operated by the Student with FlySkyI6?

human890209, 2020-06-21:
Hi,
The official FS-I6 firmware's Trainer Mode only give Channel1~4 to the Student when activated. I didn't find any info about giving more channel control rights (like 8 channels) to the student. Is there some hidden settings？
Does FlySkyI6 firmware modify this part to give more channels to the student?

Yenya, 2020-08-03:
@human890209 - FlyPlus 1.7.6 sends 8 channels from its backside PPM port, so as Student, the output is definitely here. Note that when the true Student mode is selected from the left menu, channels 5 and 6 are hardcoded to VrA and VrB, respectively, and channels 7 and 8 are inactive. The PPM output is, however, active all the time, even without switching to the Student mode from the left menu.

But as for the input, i think both stock FlySky firmware and FlyPlus firmware handle only the first four channels from the PPM input in the Teacher mode with Teacher/Student switch engaged.

Apart from that, PPM input channels 4 to 6 can be assigned to a separate Rx output channels, but this is orthogonal to the teacher/student mode.

human890209, 2020-08-04:
@Yenya Thanks. 4 channels are not enough for my project. Maybe that's kind of an old school controller style.