qba667/FlySkyI6 #105: Add more mix channels

torandi, 2020-07-11:
If possible, it would be very nice with more mix channels. In a dual motor setup, with dual steering servos. Having 4 mixes would allow differential mixing.

Maybe something that could be a build option, sacrificing something else?
(I'm a developer myself, and happy to implement it, but I'm not familiar with this code)

Yenya, 2020-07-19:
Yes, more mixes would be a nice enhancement. However, the memory inside the i6 MCU is scarce, and we have already (ab)used the memory for models #15 to 20. Also, FlyPlus is not a complete firmware rewrite, but instead binary-patched original firmware. So we would have to disassemble and rewrite the whole mixing code.

FLCostas, 2020-12-09:
Hi, Yenya. It is a great work what are you doing here.
It would also be very interesting for me to have more mixes.
I am also a developer, but I am not used to working on this type of project.
I have taken a look at the sources, but from your previous comment I understand that there is no specific module for mixing, you would have to disassemble the existing one and then try to modify it.
Anyway, if you think I could help in any way to try to make this possible, please let me know what I could do.