qba667/FlySkyI6 #108: SwB Centers Channels 1-4

Thesciman, 2020-07-20:
SwB is not configured to do anything, it has no mixes and throttle hold is not enabled. However, when I toggle it, immediately channels 1 through 4 are moved into the center, and moving the sticks has no effect. If I revert SwB to its original position, I can regain control and the sticks work as normal, moving their respective channels.

Yenya, 2020-07-23:
It looks like `SwB` is configured for teacher/student switching. Go to `System` → `Trainer mode`, and check whether `Mode` is `On` and `Switch` is `SwB`, and the third line can be switched between `Engaged` and `Not engaged` using `SwB`.

If it is, you can just set `Mode` to `Off`. Remember to save the settings by leaving that menu with long holding the `Cancel` button.

Thesciman, 2020-07-23:
I also had an issue where the rudder and elevator channels were not producing the full range of motion, even after re-calibrating the sticks many times. I was able to solve both these issues by doing a flash reset.

closed 2020-07-23 by Thesciman

unitware, 2020-07-23:
Double check that trainer/buddy mode isn’t enabled. 

Skickat från min iPhone