qba667/FlySkyI6 #110: Please help, I can not create the firmware file = (

maxtwin048, 2020-07-31:
Sorry for the stupid question, I have never done that. I don't know how to use the "make". 
it is possible to do this in Visual Studio Code or something similar? I just need to change the name of the modes in the original firmware. All day I have been looking for an answer in Google, but I didn’t understand anything = ( In what system do I need to do this WIN, Linux, Mac? I tried to load the file "makefile" in more than 10 different softwares. I entered all the commands from the description, but I received error 127. Perhaps entered in the wrong place or in the wrong order) Please, **Great Guru Yenya** !!! Please help me )))

This is what i was getting in terminal

MacBook-Pro:build userName$ make all
Building file: ../utilities/fsl_debug_console.c
Invoking: Cross ARM C Compiler
arm-none-eabi-gcc -mcpu=cortex-m0plus -mthumb -Os -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -Wall  -g -D"CPU_MKL16Z64VLH4" -I../startup -I../board -I../utilities -I../CMSIS -I../drivers -std=gnu99 -MMD -MP -MF"utilities/fsl_debug_console.d" -MT"utilities/fsl_debug_console.o" -c -o "utilities/fsl_debug_console.o" "../utilities/fsl_debug_console.c"
/bin/sh: arm-none-eabi-gcc: command not found
make: *** [utilities/fsl_debug_console.o] Error 127

Yenya, 2020-08-03:
Sorry, I have never used Visual Studio Code. There are probably tools available for Windows too - just get a make (GNU make will do) and GCC+binutils for ARM for cross-compiling.

The "arm-none-eabi-gcc: command not found" error means that you don't have GCC for ARM installed.

maxtwin048, 2020-08-04:
can not understand how to install "make"(((  Can someone be able to shoot a video how to do it? I'll be very thankful!!!

maxtwin048, 2020-08-04:
[image]
`C:\Users\User\Desktop\FlySkyI6-master\source\build>make all
'Building file: ../utilities/fsl_debug_console.c'
'Invoking: Cross ARM C Compiler'
arm-none-eabi-gcc -mcpu=cortex-m0plus -mthumb -Os -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -Wall  -g -D"CPU_MKL16Z64VLH4" -I../startup -I../board -I../utilities -I../CMSIS -I../drivers -std=gnu99 -MMD -MP -MF"utilities/fsl_debug_console.d" -MT"utilities/fsl_debug_console.o" -c -o "utilities/fsl_debug_console.o" "../utilities/fsl_debug_console.c"
process_begin: CreateProcess(NULL, arm-none-eabi-gcc -mcpu=cortex-m0plus -mthumb -Os -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -Wall -g -DCPU_MKL16Z64VLH4 -I../startup -I../board -I../utilities -I../CMSIS -I../drivers -std=gnu99 -MMD -MP -MFutilities/fsl_debug_console.d -MTutilities/fsl_debug_console.o -c -o utilities/fsl_debug_console.o ../utilities/fsl_debug_console.c, ...) failed.
make (e=2): The file cannot be found.
make: *** [utilities/subdir.mk:29: utilities/fsl_debug_console.o] Error 2
`

Yenya, 2020-08-05:
It looks similar to the error from the Mac above - you probably don't have ARM toolchain installed at all - it cannot find arm-none-eabi-gcc which it tries to invoke.

maxtwin048, 2020-08-05:
[image]

here is what I have installed

Yenya, 2020-08-05:
So maybe the name of the ARM compiler on your system is different than `arm-none-eabi-gcc`? Or the compiler is not in the shell command PATH?

maxtwin048, 2020-08-05:
**@Yenya** 

looks like magic) if you make "make" twice in a row, then the compilation goes. but I'm not sure if everything goes well?

[image]

maxtwin048, 2020-08-05:
thank you so much for your support, Great Guru **@Yenya**  !!!

Should there be a lot of "Warning" labels when compiling? In the terminal they are highlighted in pink

[image]

maxtwin048, 2020-08-06:
[image]
[image]
[image]

The compiled file weighs more than it should. the file "org.bin" in the "build" folder also occupies 64 kb, which may not be acceptable

maxtwin048, 2020-08-06:
**Please, Help ))))**

Yenya, 2020-08-19:
You have to use the *_updater*.bin files for downloading with fs-i6-updater, as described in the documentation: https://github.com/qba667/FlySkyI6/wiki/Development

closed 2020-08-19 by Yenya