qba667/FlySkyI6 #112: Help with compiling the firmware (((

maxtwin048, 2020-08-17:
Help me understand what the compilation problem is and what "Error 1" means? 

Please, Help GREAT GURUS @qba667 @Yenya @gh05t88 @unitware @hali9 

[image]

[image]

Yenya, 2020-08-19:
@maxtwin048 - it is explained in the documentation: https://github.com/qba667/FlySkyI6/wiki/Development
You made the code bigger (in the .mod_code_E140_E754 section) and it does not fit between 0xE140 and 0xE754.

Also, the Issues in this repo are not meant as a help system - they are for reporting bugs and problems. If help is needed, you can for example join the rcgroups thread mentioned in the front page.

closed 2020-08-19 by Yenya

qba667, 2020-08-19:
@maxtwin048 Probably caused by compiler, we are using version 4.9.3. This one for sure will compile not changed code.

Yenya, 2020-09-23:
FWIW, I use arm-none-eabi-gcc-cs-9.2.0-4.fc32.x86_64, arm-none-eabi-binutils-cs-2.32-3.fc32.x86_64, and arm-none-eabi-newlib-3.1.0-4.fc32.noarch. But yes, different toolchain versions can lead to different code segment sizes.