qba667/FlySkyI6 #114: Is it possible to get open source 1.7.5_fast_telemetry.zip?  Thank you very much guys!!!

maxtwin048, 2020-09-07:
Is it possible to get open source **1.7.5_fast_telemetry.zip**?  Thank you very much guys!!!  **All old firmwares are compiled**, but the latest firmware does not compile (

maxtwin048, 2020-09-14:
@Yenya Please Help 😭

maxtwin048, 2020-09-14:
@qba667 please help 😭😭😭

Yenya, 2020-09-14:
I think fast_telemetry corresponds to the commit  # 26b030248543b91bc67320bd0f50e7b59e0ba084.
Anyway, what problems do you have with compiling the current master?

maxtwin048, 2020-09-14:
@Yenya 

this is what I get while compiling the master(
[image]

maxtwin048, 2020-09-14:
but 1.7.5 compiles without problems.

but 1.7.5 compiles without problems. but telemetry is slow. I cannot find the telemetry update values ​​((( Help please @Yenya

Yenya, 2020-09-23:
As the linker says, the compiled section overlaps with the protected section from the original firmware. Maybe use a different compiler version or use -Os switch to force the compiler to make smaller code? In my compiler the section fits the assigned size. Or you can edit the linker script (MKL16Z64xxx4_flash.ld) and assign some functions to the other address blocks.

maxtwin048, 2020-09-29:
@Yenya 
[FlySkyI6-1.7.5.tar.zip](https://github.com/qba667/FlySkyI6/files/5301443/FlySkyI6-1.7.5.tar.zip)

 , could you help me, and speed up the telemetry update in this particular firmware? I will be very grateful to you) Thank you very much !!!
the original code from the repository which compiles for me without problems, below)

jan5650, 2020-11-05:
I had the same problem :(

[image]

When i compile the last firmware, i get this error...

When i use "make -i all" then it's finish, but i don't know its good?

Another Question: when its finished i get a .map file. How can i get a .bin from this? Thank you!

Yenya, 2020-11-11:
I believe the original question has been answered in the first comment, so I am closing this issue.

closed 2021-12-24 by maxtwin048