qba667/FlySkyI6 #116: Edit Firmware

jan5650, 2020-09-21:
**Dear qba667!**

You did great job! I used your 1.7.6 firmaware. I like the Fast telemetry change!

But i want to translate some of text in the firmware. I opened the .bin file with HxD, i found the text i want to edit, changed it, saved it, and when i update the firmware into my remote its get Bricked.... I can solve the Bricked remote, but i cannot edit the text in firmware to not to beign bricked. Is there any way to translate some of text in firmware?

How can i do it?

Thank you!

[image]

Yenya, 2020-09-23:
@jan5650 - there seems to be some kind of checksum attached to the firmware, so that the bootloader can verify it. Look at the Makefile to see how it is computed.

jan5650, 2020-10-01:
Thank you Yenya for the Answer!

Sadly i am lack of this knowledge. I know what is crc, but i dunno how to calculate this, and attach to the file.
Is there any program that can made it automaticly or how can i do it manually? Can yuo help me pls? Thank you!

Yenya, 2020-11-11:
As I said, look at the Makefile, how the final .bin firmware is made from the original elf. The checksum is computed there and added to the firmware file.

dremugit, 2021-07-18:
I do this by editing source\build\org.bin, then recompile on top. See http://www.slosh.com/rc/flysky/compile/