qba667/FlySkyI6 #118: Feature request: slow servo (for landing retracts / flaps)

8jigarpatel, 2020-10-26:
I am not sure if this is implemented already, but if not, it would be nice to look at the possibility of implementing slow / smooth servo movement when switch is flipped up or down. That will allow for smooth transition of flaps (using 3 position switch) or landing retracts (using 2 position switch).

8jigarpatel, 2021-02-01:
@Yenya any chance you could provide a rough ETA for this if it's feasible? I browsed through the code, but couldn't really understand what would need to change (I have C# background)

Yenya, 2021-02-01:
@8jigarpatel, I don't think it is feasible. The current mixing code is still intact from the original FW, and it is handled in assembly.

There are plans to rewrite the firmware completely, but it is outside of the scope of this project.

I will keep the issue open, though.

8jigarpatel, 2021-02-09:
@Yenya That makes sense. It's good to know that you're looking at complete re-write of the firmware.

Could I know the anticipated timeline for the project? I understand that may not be accurate if it is a community driven project, but just curious.

Also, what would be the skillset required in development of the new firmware?