qba667/FlySkyI6 #121: 1.7.6 contains error, cannot compile

jan5650, 2020-11-07:
Good day!

**When i compile the last firmware from published 1.7.6 source code i get this error...**

[image]

Older firmwares compiled with no problem... But with this 1.7.6 its not working.
I can use the modded 1.7.5., but it's IBUS speed a way slower than 1.7.6

Is there any way, to get the 1.7.6 IBUS speed with 1.7.5 firmware? What i need to change?
Or can you publish a working 1.7.6 source code?

Thank you very much!

Yenya, 2020-11-11:
So, which error do you get? Sorry, those screenshots are awful and useless. Use the text.

As I wrote in the other issues you opened, it might be that your toolchain compiles the source to a slightly bigger machine code, and it does not fit into the designated space. In that case, you will get a section overlap error from the linker. I cannot do anything about it, because the space inside the FS-i6 MCU memory is really tight, and we want to use it for features we have implemented. You can try to reorder the code in the linker script if you can find a bigger segment where to put some functions from the section which causes that linker error.

Also, the published source code fully corresponds to the published binaries, and can be compiled on my Fedora 32 system.

jan5650, 2020-11-11:
Now i get the full error:

[image]

It's because you sad: section overleap...
How can i edit linker script? where is it? anny suggestion what i need to change? Thank you!

Yenya, 2020-11-11:
The linker script is source/MKL16Z64xxx4_flash.ld, and functions/data are assigned with __attribute__(section) - see source/source/flysky.h for an example how it is done.

From the error message it looks like .mod_code_E140_E754 was compiled 5 bytes bigger than the assigned space, so you need to save those 5 bytes somehow. For the reference, my toolchain compiled this section with two bytes to spare - my fs-i6.map:

```
.mod_code_E140_E754
                0x000000000000e140      0x612
 *(.mod_customAlarmsCheck)
 .mod_customAlarmsCheck
                0x000000000000e140      0x1d8 ./source/alt.o
                0x000000000000e140                CheckCustomAlarms
 *(.mod_timerConfig)
 .mod_timerConfig
                0x000000000000e318      0x1d4 ./source/alt.o
                0x000000000000e318                TimerConfig
 *(.mod_SW_B_config)
 .mod_SW_B_config
                0x000000000000e4ec       0xd8 ./source/alt.o
                0x000000000000e4ec                SwBConfig
 *(.mod_mapSNR)
 .mod_mapSNR    0x000000000000e5c4       0x44 ./source/alt.o
                0x000000000000e5c4                mapSNR
 *(.mod_extractConfig)
 .mod_extractConfig
                0x000000000000e608       0xb4 ./source/alt.o
                0x000000000000e608                extractConfigCh7_14
                0x000000000000e660                saveAuxCh5_14
 *(.mod_createPacketCh1114)
 .mod_createPacketCh1114
                0x000000000000e6bc       0x60 ./source/alt.o
                0x000000000000e6bc                extractConfig
                0x000000000000e6c8                createPacketCh1114
 *(.mod_auxChannelsText)
 .mod_auxChannelsText
                0x000000000000e71c       0x36 ./source/alt.o
                0x000000000000e71c                auxSW

```
It ends at  0xE752 (from 0xE140 + 0x612 near the top, or from mod_auxChannelsText at the bottom 0xE71C + 0x36). You can compare it to yours fs-i6.map and see which function did your compiler make bigger than mine.

Yenya, 2020-11-11:
Also, why do I even write this? It is already documented in the wiki: https://github.com/qba667/FlySkyI6/wiki/Development

jan5650, 2020-11-11:
Yes, you have right! It's written down in wiki... 

Sadly i am not so great at developing :( I am just an amateur who want to translate some text in firmware...

So, this is what i seen in .map file:

[image]

What is it mean? How can i save 5 byte?

Yenya, 2020-11-11:
Could you _please_ use the text instead those screenshots?

jan5650, 2020-11-11:
Sorry:

.mod_code_E140_E754
                0x000000000000e140      0x61a
 *(.mod_customAlarmsCheck)
 .mod_customAlarmsCheck
                0x000000000000e140      0x1dc ./source/alt.o
                0x000000000000e140                CheckCustomAlarms
 *(.mod_timerConfig)
 .mod_timerConfig
                0x000000000000e31c      0x1cc ./source/alt.o
                0x000000000000e31c                TimerConfig
 *(.mod_SW_B_config)
 .mod_SW_B_config
                0x000000000000e4e8       0xe8 ./source/alt.o
                0x000000000000e4e8                SwBConfig
 *(.mod_mapSNR)
 .mod_mapSNR    0x000000000000e5d0       0x40 ./source/alt.o
                0x000000000000e5d0                mapSNR
 *(.mod_extractConfig)
 .mod_extractConfig
                0x000000000000e610       0xb4 ./source/alt.o
                0x000000000000e610                extractConfigCh7_14
                0x000000000000e668                saveAuxCh5_14
 *(.mod_createPacketCh1114)
 .mod_createPacketCh1114
                0x000000000000e6c4       0x60 ./source/alt.o
                0x000000000000e6c4                extractConfig
                0x000000000000e6d0                createPacketCh1114
 *(.mod_auxChannelsText)
 .mod_auxChannelsText
                0x000000000000e724       0x36 ./source/alt.o
                0x000000000000e724                auxSW

.reserved_after_code_E140_E754
                0x000000000000e754        0x4
 *(.reserved_after_code_E140_E754)
 .reserved_after_code_E140_E754
                0x000000000000e754        0x4 ./source/main.o

Yenya, 2021-02-01:
It looks like you have a different compiler than I have. For example, your TimerConfig weighs 0x1cc bytes, while mine is 0x1d4, eight bytes longer. Other routines are also different, with the total size of .mod_code_E140_E754 section is  0x61a on your compiler, and 0x612 bytes (8 bytes less) on my compiler.

fanatik911, 2021-03-27:
i have the same error while compile
`/mnt/c/111/source/build$ make all
Building target: fs-i6.elf
Invoking: Cross ARM C++ Linker
arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -Os -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -Wall  -g -T "../MKL16Z64xxx4_flash.ld" -Xlinker --gc-sections -Wl,-Map,"fs-i6.map" --specs=nano.specs -specs=nosys.specs -Xlinker -z -Xlinker muldefs -o "fs-i6.elf"  ./utilities/fsl_debug_console.o ./utilities/fsl_notifier.o ./utilities/fsl_sbrk.o ./utilities/fsl_shell.o  ./startup/startup_MKL16Z4.o ./startup/system_MKL16Z4.o  ./source/adc.o ./source/alt.o ./source/backlight.o ./source/console.o ./source/debug.o ./source/flysky.o ./source/font.o ./source/lcd.o ./source/lmul.o ./source/main.o ./source/mod.o ./source/print.o ./source/screen.o ./source/uart.o ./source/configurePins.o ./source/extraChannels.o ./source/swe.o ./source/swb.o ./source/displaySensors1.o ./source/displaySensors2.o ./source/acFrame.o ./source/getSensorName.o ./source/getSensorName2.o ./source/printTimer.o ./source/printTimer1.o ./source/printTimer2.o ./source/formatSensorValue.o ./source/formatSensorValue2.o ./source/loadSettings.o ./source/callCreatePacket1.o ./source/callCreatePacket2.o ./source/voltTelemetry.o ./drivers/fsl_adc16.o ./drivers/fsl_clock.o ./drivers/fsl_cmp.o ./drivers/fsl_common.o ./drivers/fsl_cop.o ./drivers/fsl_dac.o ./drivers/fsl_dma.o ./drivers/fsl_dmamux.o ./drivers/fsl_flash.o ./drivers/fsl_gpio.o ./drivers/fsl_i2c.o ./drivers/fsl_i2c_dma.o ./drivers/fsl_llwu.o ./drivers/fsl_lpsci.o ./drivers/fsl_lpsci_dma.o ./drivers/fsl_lptmr.o ./drivers/fsl_pit.o ./drivers/fsl_pmc.o ./drivers/fsl_rcm.o ./drivers/fsl_rtc.o ./drivers/fsl_sai.o ./drivers/fsl_sai_dma.o ./drivers/fsl_sim.o ./drivers/fsl_smc.o ./drivers/fsl_spi.o ./drivers/fsl_spi_dma.o ./drivers/fsl_tpm.o ./drivers/fsl_tsi_v4.o ./drivers/fsl_uart.o ./drivers/fsl_uart_dma.o  ./board/board.o ./board/clock_config.o ./board/pin_mux.o
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: section .reserved_after_code_E140_E754 LMA [000000000000e754,000000000000e757] overlaps section .mod_code_E140_E754 LMA [000000000000e140,000000000000e759]
collect2: error: ld returned 1 exit status
makefile:69: recipe for target 'fs-i6.elf' failed
make: *** [fs-i6.elf] Error 1`
Can you give link of compiler you use?

dremugit, 2021-04-12:
FWIW, on my Windows system, I can successfully compile with the "GNU Arm Embedded Toolchain 9-2020-q2-update" (gcc-arm-none-eabi-9-2020-q2-update-win32.exe) from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads