qba667/FlySkyI6 #122: channel 1-6, end points can only be set for the right left half

mindquest111, 2020-11-10:
in 1.7.6, i try to set up end point for channel 1-6, when i us OK to scroll to ending point setting, it only allow me to scroll (select) the left half of the end point settig, it does not scroll to the left half, which means i can only set the end point for only 1 end of each channel. is this a known problem? is there any work around？ help!

Yenya, 2020-11-11:
This is not a feature of FlyPlus, it is the same in the stock firmware: you have to move the channel input itself (i.e. the stick, switch or potentiometer) to the other half of its range - then the endpoint setup switches to the other end and you can adjust it.

mindquest111, 2020-11-11:
thank u. got it. now i feel like, duh... me

closed 2020-11-11 by mindquest111