qba667/FlySkyI6 #123: Key ->Throttle Hold, On/Off deos not seem to matter at all

mindquest111, 2020-11-23:
Key ->Throttle Hold, On/Off deos not seem to matter at all. What is the point of having throttle hold toggling on or off, if it doe not turn off throttle hold at all. Regardless if throttle hodl is turned on or off,  throttle hold always seem to only respond to the assinged swtich engaged / not engaged,

mindquest111, 2020-11-23:
Is there anyway of turning off throttle-hold off?

Yenya, 2021-02-01:
@mindquest111 - you have to set
Key → Throttle hold → Hold
to "Off", and then confirm it by long pressing Cancel. If you go to the same submenu again, the "Off" value should be visible there. As soon as it is set to Off, the throttle channel value, which can be seen at
Key → Display
should not react to the throttle hold switch at all. If it does, you might have assigned the same switch as Throttle curve as well, and what you see can be caused by the Throttle curve function instead of Throttle hold.