qba667/FlySkyI6 #124: TGY/FS-CVT01 doesn't work

zajc3w, 2020-12-06:
I have an ia6b rx with Turnigy TGY-ATM01 temperature and TGY-CVT01 battery voltage sensors,
after tx reflash i've lost voltage sensor
Only temperature works but is not selectable as displayed value in rx setup->choose sensors.

Yenya, 2021-02-01:
Do you have by chance set up the SBUS/IBUS option incorrectly? Sorry, I don't have any of these sensors myself.

Recompiling the project with newer base firmware would require disassemble the original FW first, identify "unneeded" sections, modify the linker script to reuse them for our code, and of course identify places where to hook our code at. So no, rebasing this project on top of a different FW is not an easy project.

If I remember it correctly, RSSI also depends on the firmware in the Rx.

zajc3w, 2021-02-01:
Thank for reply.
Rebating sounds like different league to my skill level.

What's  strange it that some sensors work OK passing trough not visible ones.

It must be rx firmware issue, brand new ia6c gives me both rssi and voltage feedback.
I'll have to invest in st-link and play with fw versions for ia6b,I will report back my findings here.