qba667/FlySkyI6 #127: reversed gimbals after flashing

lidorshimoni, 2021-01-30:
the gimbals (channels 1-4) are reversed, even at startup when it warns to place all switches in their up position and **throttle down**, I need to place all switches in their up position and **throttle up**! I'm currently just using the remote with all 4 channels inverted to fix it but it's still pretty annoying. did I do something wrong? am I the only one who had this problem?

8jigarpatel, 2021-02-01:
Give this a try maybe?
[https://www.youtube.com/watch?v=FkJRcjszEd8&ab_channel=-Mirek-](https://www.youtube.com/watch?v=FkJRcjszEd8&ab_channel=-Mirek-)

lidorshimoni, 2021-02-14:
> Give this a try maybe?
> https://www.youtube.com/watch?v=FkJRcjszEd8&ab_channel=-Mirek-

tried it. didn't work. the sticks are still reversed.

Yenya, 2021-06-14:
Hi, I have just flashed a newly-bought FS-i6 to FlyPlus 1.7.6 (as released), and do not see this behaviour. That is not to say your issue does not exist, but I wonder what hardware difference there can be between yours FS-i6 and at least 6 FS-i6 I have flashed so far.

lidorshimoni, 2021-06-14:
Yeah it does sound really weird. 
I guess I’m the only one and it’s not related to the firmware.
I fixed it by switching the wires of the pots.

closed 2021-06-14 by lidorshimoni