qba667/FlySkyI6 #130: compatible with tgy-i6

olivierrieux, 2021-02-26:
Hello,
I assume FlyPlus works on Turnigy TGY-i6 just like on FS-i6. Is that correct?

Cobalt6700, 2021-03-28:
I have 1.7.6 installed on a TGY-i6 and working well - they are basically the same other than the branding (I think the boards are actually the same)

olivierrieux, 2021-03-28:
Thank you for yor answer. I just happened to install it 2 days ago (when i recieved the usb adapter). Works fine untill now.
A great THANK YOU to the developpers !!!
I am just unsure whether the vario (with FS-CAT01) will be accurate enough to reliably indicate glider thermal ascention as low 1m/s (It looks that a vario sensors (with intrinsic pressure derivative aquisition) is likely to be more accurate than a numeric derivative computation of digitalized altitude, but this is definitely a sensor issue and not a transmitter issue).

closed 2021-03-28 by olivierrieux