qba667/FlySkyI6 #133: Change Airplane and Helicopter images

dremugit, 2021-04-12:
For those using the FS-i6 for projects that don't fly (trucks, trains, boats, whatever), it's possible to change the graphics displayed on the home screen. Does require recompiling as you gotta edit the org.bin.

Images are 52px wide x 15px high, stored as 104 bytes. The first set of 52 bytes is the top row of the image, going left-to-right, with bit 0 being the top of the image and bit 7 being the middle. The second set of 52 bytes is the bottom row, with bit 0 being at the middle of the image and bit 6 being the bottom. Bit 7 is unused. A "1" bit shows up as on (black) on the LCD and "0" as off.

The Helicopter image is stored at 0xD5F9 through 0xD660, and the Airplane at 0xD661 through 0xD62D (again, in org.bin.) 

As my Python-fu is very weak, I did up a Perl to patch a 52x16x2 BMP into the bin.

[image]
[image]
[image]

Also, @qba667 or @Yenya or other devs, I have some other stuff to contribute, don't know how best to do it. There's a little Arduino project to combine switches and/or pots to the PPM in, gives up to six additional switches / three additional pots

[image]

If you'd rather not clutter this page with Arduino stuff I can put it elsewhere, maybe link to it from this issue?

The hardware mod for adding a pot to SwC could maybe use a short writeup (and you can change its name to "VrC", which leads into the next thing.)

I'd also be happy to do a more complete set of instructions to compile. You might consider it too risky, as there's always the risk of folks bricking their TX's, but it would save having to do multiple builds (no startup checks, different images as above, etc etc.) Anyway, a thought.

8jigarpatel, 2021-07-08:
I like this bit: "I'd also be happy to do a more complete set of instructions to compile."
I have coding background in C# (not C tbh), I won't lie I skimmed through the files here and they didn't make a lot of sense to me, But I have spare fs-i6 that I can happily experiment with. 

I specifically want to explore possibility of adding a feature that would allow for servo moments to be slowed down (for flaps / landing gears). I did raise a feature request for this (https://github.com/qba667/FlySkyI6/issues/118), and Yenya mentioned it is not feasible. But if I knew hot to navigate through project / compile etc., I am willing to spend some time on it :)

Also, what are your thoughts on the feature request, and possibility of implementing it?

dremugit, 2021-07-08:
Wrt compiling, I actually did up a howto in the interest of changing the airplane and helicopter graphics:
Compiling the QBA667 firmware for FlySky FSi6 and modifications

| 
| 
|  | 
Compiling the QBA667 firmware for FlySky FSi6 and modifications

 |

 |

 |

Wrt the feature req, the problem is that as I overstand it, thism firmware is more the binary with patch deltas on top of it, so it's more like you compile the deltas, not the whole thing from source. We therefore don't have access to change the entire thing. Plus, while it seems easy (and is, sort of, explore in a moment) I can see where it would be hard to add onto the existing code. Which servo(s) to control, which to not? At what speed(s)? Etc.
The developers of this firmware have said that they've basically maxed out the memory space available (in fact, depending on the compiler used, you can easily go over the 64k or whatever it is), plus I believe they've moved on to doing updates for newer transmitters. I'm continually amazed by how much they squeezed out of the FSi6 as it is :)

Now, there are other solutions to give you slower servo movements. My use case is RC *ground* vehicles, cars and trucks and construction equipment and such, so I have more space and don't care so much about weight as y'all flying-thing people. Point being that you can very easily stick an Arduino in between the RX and the servo(s). It reads the iBus data, and then controls the servo(s) any way you want. It would act much like a servo reverser (or a servo scaler, though I think those aren't much a thing any more.) Basically the code just says e.g. "if iBus channel X value is [more | less] than servo X value AND more than 10mS has passed, [increment | decrement] the servo value one" (obvi 10mS is arbitrary as is the inc/dec value.) This then slews the servo at the desired rate. There are also fancy Arduino servo libraries with slew rate as a function if you don't want to DIY.
Depending on the scale of your aircraft, the Arduino may or may not work for you. They can get tiny; the Nano would be my choice for this, ~35x15mm and 3-4g depending on connectors soldered, so unless your craft is tiny, might be doable. (They're also cheap, couple USD per, so I've used them as servo reversers as well as speed control.)
If that doesn't fit, you may or may not know that there are digitally programmable servos. I know the Hitecs have an option to change slew rate over a wide range; I used this on a car with popup headlights so they go foooooooooooooooooooooooop nice and smooth, instead of FOOP up and FOOP down. :)
My .02. Hope that's helpful, with inflation and the rate of the USD, worth the electrons it's printed on if not less, etc. Contact info at the end of the web page above if you wanna ping me offline for anything.

-- A