qba667/FlySkyI6 #134: No SwE in Functions -> Switches assign, double tone when exiting "Extra" menu, turning mixes on/off with a switch

awocrf, 2021-06-05:
1. SwE is not possible to choose in Functions -> Switches assign.
2. There are 2 tones played when exiting "Extras" (High and low).
Also settings that aren't different from stock firmware have low tone and those modified have high tone when exiting. Is this intensional?
3. Can you add possibility to choose a switch that would be turning given mix on and off?

dremugit, 2021-07-18:
I think for #1, you used a firmware that didn't include SwE. Be sure to burn a BIN that says "SWE" in the filename, e.g. fs-i6-swe_updater_05_27_16_45.bin." If you just use e.g. "fs-i6_updater_05_07_12_30.bin", you won't get SwE. (Different code is required because if SwE not wired up, the input floats and the processor goes crazy trying to track it.)

I think for #5, you're asking about https://github.com/qba667/FlySkyI6/issues/21 (or https://github.com/qba667/FlySkyI6/issues/74, dupe)

Can't speak to #2-3-4.

HTH.

-- A

awocrf, 2021-07-18:
1. Actually I am using SwE version.
5. Yeah that's a duplicate of these.

dremugit, 2021-07-18:
Wrt #1, doh, sorry. Occam's Razor, assume the simplest cause, right? =))

Another possibility: note that SwE is out of order in the options, like it's the very LAST item, after PPM1/2/3, so maybe you just didn't go far enough?

-- A

awocrf, 2021-07-18:
I think I scrolled to the end but I'll check it later again because right now it's 1:15 am and I would wake up everyone with radio beeping.

awocrf, 2021-07-19:
In switches assign you can only choose SwA, B and D.

dremugit, 2021-07-19:
Doh! My bad. I was thinking you meant Menu, Functions Setup, *Aux Channels* ... but you mean Menu, Functions Setup, Switches Assign (where it clearly says "functions assign".)

So yeah, you're right, "switches assign" only allows some switches.

-- A