qba667/FlySkyI6 #135: 1.7.5 fast telemetry

jan5650, 2021-06-14:
**Hi!**

Do you added fast telemetry update for 1.7.5 version. (named 1.7.5_Fast_telemetry.zip)
Sadly i cannot compaile the 1.7.6 version, because complaier mismatch, so i need to use the 1.7.5 version.

**My questin**: is there any changes in **Ibus Telemetry speed** between 1.7.5 fast telemetry and 1.7.6 ?

Yenya, 2021-06-14:
@jan5650  - you can do "git log" and display all the commits between 1.7.5 and HEAD. Some of them should describe the fast telemetry update.

Also, can you test whether the current HEAD (863819f) compiles with your toolchain?

jan5650, 2021-06-14:
"Also, can you test whether the current HEAD (863819f) compiles with your toolchain?"

Maybe if I Reorder linking as you mentonied then i can compile the 1.7.6 with newer GCC?
That would be great!

jan5650, 2021-06-15:
Dear Yenya!

With moving .mod_auxChannelsText to the very end of the code in the firmware lines it's work Great!!
Now i can compile the latest 1.7.6. firmware with no problem!

Thank you!

closed 2021-06-16 by jan5650