qba667/FlySkyI6 #136: MxAl with CAT01

lpetya, 2021-06-22:
A huge Thank you for the developers for the mod.

I have a question:
I can't find the MxAl - maximum altitude - at the sensors list.
I use TGY-i6 transmitter with FlyPlus 1.7.6 firmware and TGY-iA6B receiver with FS-CAT01 altitude sensor.
All telemetry data received correctly, altitude too, of course.
Can I receive the max alt with this configuration? How?

In Sensors list I find these items only:
IntV  1
Temp  2
Press.2
Tx V  1
Alt   2
SNR   1
Noi   1
RSSI  1
Err.  1

There is no MxAl. I'm missing...

Please help me.
Thanks

Yenya, 2021-06-22:
@lpetya: the MxAl value is not sent by FS-CAT01. MxAl is something I implemented in my own Arduino-based altitude sensor: https://github.com/Yenya/ibus-altitude-sensor

lpetya, 2021-06-22:
Thank you for the fast reply.

I thought this is a calculated value like Alt from the press and temp.

All in all, this is a good job!

closed 2021-06-22 by lpetya