qba667/FlySkyI6 #154: Replace Timer on Screen

jan5650, 2022-04-27:
Hello!

I am enjoy using this firmware. Helps me great!

Is there any way to remove timer from the upper left section of screen (00:00:00)?
I know its hardcoded, but dunno wich function i need to modify.
I found two:

**Sources\Mod.h**
[image]

_Can i just simply write a short string without "00:00:00" like "FlySky" ? I think not, but worth a question :D_

**Sources\Alt.c**
[image]

What happend, if i uncomment the whole content of this function? There will be a blank space on the upper-left?
What i need to write there if i want "FLYSKY" text on the upper left section?

Thank you!

Yenya, 2022-04-27:
I think it should work  - change the string "00:00:00" to "FLYSKY  " and comment out the body of printTimer() except the last two lines.

jan5650, 2022-04-27:
Thank you for the fast answare! 
I will try it, and post the result.