qba667/FlySkyI6 #161: Increase endpoint range

batata004, 2023-07-21:
Hi,

This firmware is awesome but there is only one thing I cant do: is it possible for you to increase the ENDPOINT range? So far I can go up to +120% and -120%. I would like to use something like 500% or even 600%. There reason is valid: I want to control a MOSFET with PWM but the duty cycle of most radios is around 10% (pulses goes from 0.5ms to 2.5ms every 20ms). If you allowed to increase the ENDPOINT to larger values, then many people could use their own MOSFETs to control anything using Flyskyi6 radio. I know I could use brushed ESCs but I will be stuck with the ESCs MOSFET and I would like to use my own MOSFETs for specific loads/voltages/currents...

Thank you and congratulations for the work you put into this firmware, it's awesome!!!

batata004, 2023-08-23:
Is there any other place where I should post this?

Yenya, 2023-09-01:
@batata004: I don't think it is possible at all. FS-i6 transmits digital AFHDS2A signal, which is transformed to PPM only at the RX side. The only PPM signal generated by the TX itself is the one comming from the mini-DIN connector on the back side of the TX.

Also, bear in mind that our firmware is not a complete firmware rewrite, but instead a binary patch to the partly reverse-engineered original FlySky firmware, with several functions and menus replaced and new features implemented only by using several memory regions we consider unused. So I doubt something like this would be possible without reverse-engineering and rewriting more parts of the firmware.

rhkavli-nautrik, 2024-09-08:
The easiest way to do this, is to use a secondary microcontroller, reading the iBus and converting the channel value to PWM using onboard timers. I've done this successfully for another project.