qba667/FlySkyI6 #163: only 14 model memory slots available

Walker555, 2023-08-21:
I have used the version 1.7.6 since 2021. As usual I have created new models by copying existing one. So far so good. Suddenly I have realized that I can't copy anymore and the list of models is limited to 14. The last copy operation also destroyed all Mixes configurations of all models. I decided to "reset" completely. I have flashed the version 1.7.6 again. It didn't help. I see 14 models FlySky01 - 14, that is all. Is this a known issue? How can I get back 20 model memory slots?

Yenya, 2023-08-21:
@Walker555 - 1.7.6 supports only 14 models. We have used the memory space previously used by  the remaining 6 models for data for additional channels (their endpoints, trim values, etc.).

Walker555, 2023-08-21:
@Yenya OK, thank you for the explanation.

closed 2023-09-01 by Yenya