qba667/FlySkyI6 #168: Error flashing firmware : Read timeout

Honza316, 2024-03-06:
I am trying to flash my fs-i6 with latest firmware 1.7.6. and I always get error flashing firmware: Read timeout at the same point 35,78 % .  I am albe to flash my fs-i6 with official firmware or with alternative 10 channel firmware. So there should not be issue with fs-i6 or with cable. Any ide what can be wrong?
Thanks a lot for any kind of help.
regards Jan

Yenya, 2024-03-06:
Hi Jan,

sorry, I haven't seen this so far. How old is your FS-i6? Could it be that FlySky has changed something in the hardware or bootloader which makes the FlyPlus installation fail? Or you can have corrupted firmware file (unlikely, but worth checking): the SHA1 checksum of the firmware file is

bfd921c2aa627018c29b1540bb42ad0b86a63b9f  fs-i6_updater_05_07_12_30.bin

Sorry, I don't have any meaningful solution for this.

-Yenya

Honza316, 2024-03-06:
Hi, thanks for quick reply. It is new FS-i6. I have just bought it. SHA-1 is correct. As I said before I can flash it with older version FlyPlus beta from 20/01/16.  I have also tryied it from another pc, but it was same. I am using Arduino for flashing it.
thanks regards Jan. 

this is what I always get
0 B / 58.00 KB [---------------------------------------------------------------------------------------------]   0.00 % 1.25 KB / 58.00 KB [=>------------------------------------------------------------------------------------]   2.16 % 9s 2.50 KB / 58.00 KB [===>----------------------------------------------------------------------------------]   4.31 % 8s 4.00 KB / 58.00 KB [=====>--------------------------------------------------------------------------------]   6.90 % 8s 5.25 KB / 58.00 KB [=======>------------------------------------------------------------------------------]   9.05 % 8s 6.75 KB / 58.00 KB [==========>---------------------------------------------------------------------------]  11.64 % 7s 8.00 KB / 58.00 KB [===========>--------------------------------------------------------------------------]  13.79 % 7s 9.50 KB / 58.00 KB [==============>-----------------------------------------------------------------------]  16.38 % 7s 11.00 KB / 58.00 KB [================>--------------------------------------------------------------------]  18.97 % 6s 12.25 KB / 58.00 KB [=================>-------------------------------------------------------------------]  21.12 % 6s 13.75 KB / 58.00 KB [====================>----------------------------------------------------------------]  23.71 % 6s 15.00 KB / 58.00 KB [=====================>---------------------------------------------------------------]  25.86 % 6s 16.50 KB / 58.00 KB [========================>------------------------------------------------------------]  28.45 % 6s 18.00 KB / 58.00 KB [==========================>----------------------------------------------------------]  31.04 % 5s 19.25 KB / 58.00 KB [============================>--------------------------------------------------------]  33.19 % 5s 20.75 KB / 58.00 KB [==============================>------------------------------------------------------]  35.78 % 5s Error flashing firmware: Read timeout.

Honza316, 2024-03-18:
I have finally successfully upgrade my fly-sky to this firmware. I used cable with FT232RL board instead of Arduino and issue with read timeout did not appear any more. 
Jan.

closed 2024-03-18 by Honza316

Yenya, 2024-03-19:
Thanks for letting us know! Could it be the baud rate of of a serial port or something?

Honza316, 2024-03-19:
Hi, 
It is hard to say I was using same setup with Arduino for flashing my Fly-sky with official firmware and it worked fine. Same for non-official 10 channel firmware. Both of them use different flashing software. So I do not know if it set up serial port different way or not. Anyway I was using manual setting on windows to 115200. 
Regards Jan.