qba667/FlySkyI6 #171: hi

oldcityboy, 2024-12-06:
Hello, FSI6 is not loading, I tried it with the original installation cable.
I also tried it with FT232RL card.

oldcityboy, 2025-01-01:
> You can use any Serial to USB device to flash the remote.
> You can use Male jumper cables to stick in the connector (including the
> ground).
>
> I've tested this recently and successfully flashed the new firmware.

I installed version 1.7.5 and I want to measure distance for my rc boat.
There are sensors on the remote control but I can't get data from the GPS.
Can you help me about this issue?

Lachezar Lechev ***@***.***>, 30 Ara 2024 Pzt, 21:21
tarihinde şunu yazdı:

elpiel, 2025-01-01:
I consider this off-topic but you can always give more details to people in the https://aerorust.org/ community on discord.

oldcityboy, 2025-01-08:
Hello, your remote control is fsi6. I could not install fsi6 1.7.6
I was able to install 1.7.5, do you have a screenshot of the 1.7.6 sensor
list?
I wanted to see and compare the difference between 1.7.5 and 1.7.6
tanks My remote control is fsi6b

Lachezar Lechev ***@***.***>, 1 Oca 2025 Çar, 21:25 tarihinde
şunu yazdı:

elpiel, 2025-01-08:
[image]

oldcityboy, 2025-01-08:
when I connect the cables to the computer, 1.7.6 does not see the program

I will try again
Thank you,

Lachezar Lechev ***@***.***>, 8 Oca 2025 Çar, 19:31 tarihinde
şunu yazdı:

oldcityboy, 2025-01-08:
Hello, I tried again, the program does not connect to the remote control
with FT232 and original flysky cable.
I connected the FT232 USB device with three cables, gnd tx rx, but it does
not connect to the program.
Is it the same program that you downloade
thanks again

Cengiz Can ***@***.***>, 8 Oca 2025 Çar, 19:35 tarihinde şunu
yazdı:

RI662, 2025-01-12:
Hello, I have the same problem.
I have a FLYSKY FS-I6X.  I tried all listed procedures without sucess.
When I connect to the computer with the cable supplied, I can update the version with the current FLYSKY software, but the current file is fixed. (1.0.52.1442). The old update software does not run and so I cannot load any version. No COM port is displayed in the device manager. Could it be that the process is different for new systems?
Regards Richard

elpiel, 2025-01-12:
Fs-i6x is not supported, as far as I know, by this firmware

oldcityboy, 2025-01-12:
I am using flysky fsi6 1.7.6 I could not install the software and I could
not connect the program with the remote control.
USB connects to the computer but I could not connect to the remote control
program
I have previously installed 1.7.5

Lachezar Lechev ***@***.***>, 12 Oca 2025 Paz, 21:21
tarihinde şunu yazdı: